The-OpenROAD-Project/yosys #1: SDC Support status

tajayi, 2019-06-17:
I never really go clarification on the SDC issues (lost the thread in the new fork). I was observing a lot of discrepancies between what was in the README, what was in the sample scripts and what was actually working. From my recollection.

It appears "abc" may support sdc but not "phys_abc"? Is this correct? The current instructions to run phys_abc no longer has sdc constraints as an input, however I still need to provide it a clock pin.

MarinaHesham, 2019-06-17:
First, I apologize for causing that confusion,
SDC is supported in both abc and phys_abc, The only method to enable it was through the abc script as shown in the second snippet in the ReadMe,
But I enabled using it through -constr option in Yosys (I guess this is more clean and easy),
Generally, Yosys itself doesn't use the SDC file, It just passes it to ABC, so the SDC parser can be enabled by one of two methods:
1.Through Yosys by adding the -s flag after the -constr flag in abc or phys_abc command, and Yosys will pass it to ABC.
2.Through ABC directly by adding -s flag to read_constr command in the ABC script.
If -s flag was not added, the typical constraints parser of ABC (non SDC) will be used.
I Modified the ReadMe and the snippets based on that modification and let me know if this is still unclear,
I also Modified the Readme for -constr option with Physical synthesis(phys_abc command), I am sorry I just missed it, 
This is mandatory for the Physical Sythesis flow, as RePlAce needs it to run in the timing driven mode, and I must run RePlAce in timing driven mode to be able to generate the SPEF that is used later.

tajayi, 2019-06-20:
Okay.. marking this as closed for now.
Thanks

closed 2019-06-20 by tajayi