The-OpenROAD-Project/yosys #2: Physical Synthesis issues

tajayi, 2019-06-17:
Using the latest version of yosys. I'm running into the following error

```
8. Executing Physical ABC pass (Physical Synthesis using ABC + RePlAce).

8.1. Extracting gate netlist of module `\gcd' to `<abc-temp-dir>/input.blif'..
rm -rf ./results/gcd/phys_abc/etc/gcd/net/trenton/x/ajayi/projects/OpenROAD/CI/build/export/OpenROAD-2019-06-17_06-36/bin/defgenerator -lef /net/trenton/x/ajayi/projects/OpenROAD/CI/test/t7_phySynth/./objects/gcd/cadre_tech_macro.lef -lib /net/trenton/x/ajayi/projects/OpenROAD/CI/test/t7_phySynth/./objects/gcd/merged.lib -verilog _tmp_yosys-abc-ic8cFC/netlist.v -defDbu 2000 -dieAreaInMicron 0 0 148 148 -siteName sc12_cln65lp -design netlist -def _tmp_yosys-abc-ic8cFC/gcd.def
argc and argv list: 
0 /net/trenton/x/ajayi/projects/OpenROAD/CI/build/export/OpenROAD-2019-06-17_06-36/bin/defgenerator
1 -lef
2 /net/trenton/x/ajayi/projects/OpenROAD/CI/test/t7_phySynth/./objects/gcd/cadre_tech_macro.lef
3 -lib
4 /net/trenton/x/ajayi/projects/OpenROAD/CI/test/t7_phySynth/./objects/gcd/merged.lib
5 -verilog
6 _tmp_yosys-abc-ic8cFC/netlist.v
7 -defDbu
8 2000
9 -dieAreaInMicron
10 0
11 0
12 148
13 148
14 -siteName
15 sc12_cln65lp
16 -design
17 netlist
18 -def
19 _tmp_yosys-abc-ic8cFC/gcd.def
LEF Parsing Done
DEF DBU: 2000
Site Found: sc12_cln65lp 400 4800
Trimmed New DieArea: 0 0 148 148.8
740 61
Executing OpenSTA to parse Verilog....
Error: TCL init script: invalid command name "version conflict for package "Tcl": have 8.5.19, need exactly 8.5.7
    while executing
"package require -exact Tcl 8.5.7"
    (file "/usr/share/tcl8.5/init.tcl" line 20)
    invoked from within
"source [file join $tcl_library init.tcl]"".
       Try deleting app/TclInitVar.cc and rebuilding.
python /net/trenton/x/ajayi/projects/OpenROAD/CI/build/export/OpenROAD-2019-06-17_06-36/share/yosys/pins_placer.py -def _tmp_yosys-abc-ic8cFC/gcd.def -output _tmp_yosys-abc-ic8cFC/gcd.def
  File "/net/trenton/x/ajayi/projects/OpenROAD/CI/build/export/OpenROAD-2019-06-17_06-36/share/yosys/pins_placer.py", line 50
    y = 0
        ^
TabError: inconsistent use of tabs and spaces in indentation
ERROR: System call "python /net/trenton/x/ajayi/projects/OpenROAD/CI/build/export/OpenROAD-2019-06-17_06-36/share/yosys/pins_placer.py -def _tmp_yosys-abc-ic8cFC/gcd.def -output _tmp_yosys-abc-ic8cFC/gcd.def
" failed!
make: *** [results/gcd/gcd.synth.v] Error 1
```

- I am able to run yosys (without phys_abc) and replace without issues so I'm not sure where the strict tcl requirement is coming from. A strict requirement on a minor tcl version would be a bit painful for the release packages.

- On the python issue, instead of calling "python myscript.py", can you shebang it (add  "#!/usr/bin/env python3" to the top of the file and "chmod +x")? that way the system can use the appropriate version of python. 

Thanks

MarinaHesham, 2019-06-17:
- I fixed the python issue, and you could find it on the Repo right now
- For the tcl, I guess it is related to openSTA as the Error says "Executing OpenSTA to parse Verilog.... "

tajayi, 2019-06-18:
So the issue was in defgenerator not replace. not sure why it broke but I'm able to work around it for now.

I'm now in the section where it's running the python scripts. At least the pins_placer.py and util.py have several issues. They are all related to indentation issues.

```
TabError: inconsistent use of tabs and spaces in indentation
```

MarinaHesham, 2019-06-18:
It seems like an issue with tabs, that's why its inconsistent (working on my machine and not yours), I removed all tabs and replaced them with spaces, Let me know if you still have the same issue

tajayi, 2019-06-20:
Okay. More progress.. 

Replace is now throwing this error
```
total_filler_area: : 504684
INFO:  FillerCell's X = 5.020195 , FillerCell's Y = 9.000000
INFO:  #CELL = 27386 =  16216 (#MODULE) + 11170 (#FILLER)
INFO:  Average Module Width = 5.392452, Row Height = 9.000000
INFO:  D_MSH = 1024 
INFO:  MSH(X, Y) = (32, 32)
INFO:  WRITE BOOKSHELF..../results/aes_cipher_top/phys_abc/etc/aes_cipher_top/experiment000/tiers/0
ERROR: n16345 have layer as HVT, but not exists in LEF
cp: cannot stat `./results/aes_cipher_top/phys_abc/etc/aes_cipher_top/experiment000/*.spef': No such file or directory
ERROR: System call "cp ./results/aes_cipher_top/phys_abc/etc/aes_cipher_top/experiment000/*.spef spef_output/netlist.spef" failed!
make: *** [results/aes_cipher_top/aes_cipher_top.synth.v] Error 1
```
I noticed defgenerator is putting pins on an invalid "HVT" layer. The layer does exist in the tech lef but ti's an Implant layer (not a routing layer).

perhaps instead of running your place_pins.py you can call ioPlacer (https://github.com/The-OpenROAD-Project/ioPlacer) which is ready.

Another alternative is to generate a file with the pin locations and pass it to defgenerator using pinLocation.

MarinaHesham, 2019-06-20:
The script just places the pins at a certain location on the same layer on
which the defgenerator puts pins which is M2 by default,
I am wondering where did HVT come from?
I will try to investigate, If you could send me that generated def (found
at the tmp directory of yosys) that would be helpful

tajayi, 2019-06-20:
HVT is the second layer that shows up in the lef so I'm guessing defgenerator just arbitrarily chose that. But replace doesn't like it

uploaded to t7_phySynth_2019-06-19_21-15.tar.gz

MarinaHesham, 2019-06-20:
I added an option '-layer' for phys_abc to define the layer number, this should be an integer value, If it is not set then the default value 2 will be used

tajayi, 2019-06-20:
Wouldn't it make sense to set it to a layer name instead? The invalid HVT is really not a routing layer but is indexed as 2. If you give the user the option to pick a layer, they may think 2 = M2

MarinaHesham, 2019-06-20:
Yes I got your point, I don't know why is this is considered an index though,
If the input layer num = 2 (which is the default), the output def will define the pin as

MarinaHesham, 2019-06-20:
Yes I got your point, I don't know why this is considered an index though,
If the input layer num = 2 (which is the default), the output def will define the pin as 
- x + NET x + DIRECTION INPUT + USE SIGNAL
 **+ LAYER M2**  ( -100 0 ) ( 100 1040 ) + FIXED ( 0 236 ) N ;
This is the output that the pins placer write so why should this mean the layer of the second index!

I also checked ioPlacer and I found that it takes the layer number as an input not a name, 
does this make sense ?

tajayi, 2019-06-20:
> I added an option '-layer' for phys_abc to define the layer number, this should be an integer value, If it is not set then the default value 2 will be used

I think I misunderstood this. Yeah, I'm happy if "-layer 2" means "-layer M2" (similar to ioPlacer).
I thought "-layer 2" would equate to the second layer found in the lef which happens to be "HVT" in tsmc65lp

MarinaHesham, 2019-06-20:
The point is that before it used to write only M2 all the time, That's why I don't know why the output was HVT, but anyway if the defgenerator used to take it as an index for any reason, The pins placer will fix that now.

tajayi, 2019-06-20:
So defgenerator usually takes the pin placement as an input arguement. My guess is that it defaults to something random if you don't give it one. It just happens to be a bad layer for tsmc65lp lef (perhaps it was M2 for your tech lef). My guess is that it lazily picked the second *arbitrary* layer in the lef.

Thanks

tajayi, 2019-06-23:
Physical synthesis is now able to complete. Thanks

closed 2019-06-23 by tajayi