The-OpenROAD-Project/yosys #3: Yosys does not handle memory macros

tajayi, 2019-06-20:
This is another issue that got lost in the new repository fork.

Yosys doesn't not appear to handle the macros. I observe the following error 
```
Used module:                     \and_op
Used module:             \bsg_two_fifo_width_p76
Used module:                 \bsg_mem_1r1w_width_p76_els_p2_read_write_same_addr_p0
Used module:                     \bsg_mem_1r1w_synth_width_p76_els_p2_read_write_same_addr_p0_harden_p0
Used module:                         \mux
ERROR: Module `\tsmc65lp_1rf_lg10_w32_all' referenced in module `\bsg_mem_1rw_sync_width_p32_els_p1024' in cell `\macro.mem' is not part of the design.
make: *** [results/bsg_manycore_tile/bsg_manycore_tile.synth.v] Error 1
```

I have ensured that the cell is in the merged lib

MarinaHesham, 2019-06-20:
If you use the usual Yosys script,
1. you first read the verilog --> No problem should arise
2. you check the hierarchy --> This is where this error happens and at this point Yosys haven't read the library file yet to recognise the macro,
Unfortunately I don't have the library file to reproduce the issue, 
But a possible solution is to use the flag -lib while reading the verilog, this should create empty blackbox modules, and then when you check, this macro is treated as a black box.
Let me know if this doesn't work.

tajayi, 2019-06-20:
Stumbled on this post: https://www.reddit.com/r/yosys/comments/2at8s7/structural_verilog_input/
I'm doing method 1. I would have preferred 4 bur ran into this issue:https://github.com/YosysHQ/yosys/issues/825

```
hierarchy -generate tsmc65lp_* o:Q o:QA o:QB \
                               i:CLK i:CLKA i:CLKB \
                               i:CEN i:CENA i:CENB \
                               i:GWEN \
                               i:A i:AA i:AB \
                               i:D i:DB i:WEN \
                               i:STOV \
                               i:EMA i:EMAA i:EMAB \
                               i:EMAW i:EMAS i:RET1N
```

This seems to work but then fails on the phys_abc command
```
8.1. Extracting gate netlist of module `\bsg_manycore_tile' to `<abc-temp-dir>/input.blif'..
ERROR: Assert `sig_d.size() == 1' failed in passes/techmap/phys_abc.cc:669.
make: *** [results/bsg_manycore_tile/bsg_manycore_tile.synth.v] Error 1
[ajayi@vlsipool-j12 t7_synth]$ 
```
t7_synth_2019-06-20_03-56.tar.gz uploaded

MarinaHesham, 2019-06-20:
Yes, phys_abc doesn't handle Memory blocks with more than one input, I will work on it and let you know.

MarinaHesham, 2019-06-24:
Now phys_abc should be able to handle Memory Blocks, I tested it, Let me know if you still have any problems

MarinaHesham, 2019-06-24:
Forgot to mention that for the memory blocks, there is multiple clock ports, so all of them should be defined by -clk_port name1 -clk_port name2 ....

tajayi, 2019-06-27:
Still have a problem with phys_abc and macros

```
ABC: Scl_LibertyReadGenlib() skipped sequential cell "SDFFYQ_X3M_A12TR".
ABC: Scl_LibertyReadGenlib() skipped sequential cell "SDFFYQ_X4M_A12TR".
ABC: Library "tscm65lp_merged" from "/net/trenton/x/ajayi/projects/OpenROAD/CI/test/t7_yosys/./objects/aes_cipher_top/merged.lib" has 643 cells (281 skipped: 228 seq; 9 tri-state; 44 no func; 26 dont_use).  Time =     1.36 sec
ABC: Memory =   78.34 MB. Time =     1.36 sec
ABC: Warning: Detected 32 multi-output gates (for example, "ADDFCIN_X1M_A12TR").
ABC: + read_blif <abc-temp-dir>/input.blif 
ABC: ** cmd error: aborting 'source <abc-temp-dir>/abc.script'
ABC: Line 31377: Cannot find gate "DFFQ_X1M_A12TR" in the library.
ABC: Reading network from file has failed.
ERROR: Can't open ABC output file `_tmp_yosys-abc-0Put38/output.blif'.
make: *** [results/aes_cipher_top/synth.v] Error 1
```

If i take out phys_abc on line 45 of this file https://github.com/The-OpenROAD-Project/release/blob/master/test/t7_yosys/scripts/synth.tcl#L45, yosys runs to completion.

uploaded t7_yosys_2019-06-26_21-14.tar.gz

MarinaHesham, 2019-06-27:
The reason is that Yosys parsed the Flip Flops to ABC but ABC knows nothing about Sequential cells that's why the error says, Can't find gate "DFFQ_X1M_A12TR" in the library.

Yosys recognises the Sequential Cells by their clock port and extract them before sending to ABC.
So this means that not all clock ports were defined.

I see in the synth.tcl script that you defined only one clock port but I noticed that the clock port for DFF (CK) is different from that connected to the Macros (CLK, CLKA)
I tried running it with defining all clk ports  and Yosys did it to completion So can you make sure that you defined all the clock ports by -clk_port CK -clk_port CLK -clk_port ..... ?

tajayi, 2019-06-27:
Hmmm.. this seems like a lot of work for the user (going through all standard cells and macros) across all designs to figure out the clock port names. It should be easily traceable from the top level clock. Is there a query that can be run in yosys to get these pin names?

For now, I've updated the scripts to include all the pins I can find. Vanilla bean now completes

closed 2019-06-27 by tajayi