The-OpenROAD-Project/yosys #4: Physical Synthesis fails for designs with macros

tajayi, 2019-06-27:
Physical synthesis fails for designs with macros. Specifically, replace is failing to run.

```
Warning: _tmp_yosys-abc-VCVMa5/netlist.v, line 44425 instance gg14480 port QA not found.
Warning: _tmp_yosys-abc-VCVMa5/netlist.v, line 44425 instance gg14480 port QA not found.
Warning: _tmp_yosys-abc-VCVMa5/netlist.v, line 44425 instance gg14480 port QA not found.
INFO:  Successfully linked: netlist
Warning: design.sdc, 9 port 'clk' not found.
NO PATH !
INFO: Total 0 worst path!
ERROR: System call "/net/trenton/x/ajayi/projects/OpenROAD/CI/build/export/OpenROAD-2019-06-27_10-41/bin/RePlAce -bmflag etc -def _tmp_yosys-abc-VCVMa5/bsg_manycore_tile.def -lef /net/trenton/x/ajayi/projects/OpenROAD/CI/test/t7_phyYosys/../testcases/tsmc65lp/tech/tech_macro_padded.lef -verilog _tmp_yosys-abc-VCVMa5/netlist.v -lib /net/trenton/x/ajayi/projects/OpenROAD/CI/test/t7_phyYosys/./objects/bsg_manycore_tile/merged.lib -sdc /net/trenton/x/ajayi/projects/OpenROAD/CI/test/t7_phyYosys/../testcases/tsmc65lp/bsg_manycore_tile/design.sdc -output ./results/bsg_manycore_tile/phys_abc -t 1 -dpflag NTU4 -dploc /net/trenton/x/ajayi/projects/OpenROAD/CI/build/export/OpenROAD-2019-06-27_10-41/bin/ntuplace4h -onlyGP -unitY 864 -capPerMicron 2.35e-13 -resPerMicron 1.5985 -timing" failed!
make: *** [results/bsg_manycore_tile/synth.v] Error 1
```

I investigated a little bit, And running the system call on the terminal results in a "Segmentation fault" from Replace. Above that there are several Errors and warnings from replace. My guess is 1) perhaps replace doesn't like defs with macros that are not placed 2) replace doesn't like the way the lib was merged.

I've uploaded t7_phyYosys_2019-06-27_13-27.tar.gz

tajayi, 2019-06-27:
Actually all physical synthesis breaks now. Not quite sure what the difference is.

I've uploaded the simple gcd design at t7_phyYosys_2019-06-27_13-51.tar.gz

MarinaHesham, 2019-06-27:
Apparently, RePlace Run didn't succeed, If you opened  t7_phyYosys_2019-06-27_13-51/results/gcd/phys_abc/etc/gcd/experiment000/gcd_final.def, you will find all cells placed at ( -2147483648 -2147483648 ), and the spef at  t7_phyYosys_2019-06-27_13-51/spef_output/netlist.spef has negative values

MarinaHesham, 2019-06-27:
This have been happening for GCD but not AES, LDPC, JPEG, I am not sure what's the problem with RePlAce.

The change that I did was running Global Placement inside Yosys instead of Detailed Placement because this gives better results and Extraction after DP don't seem to be totally right to me (Range of wire capacitance was large) I am not sure why is RePlAce Crashing for GCD but I will try to investigate it and I will let you know

MarinaHesham, 2019-06-28:
It seems like RePlAce didn't like the pins placement for some reason (which I don't know), I checked that all pins placement are inside the die bounds, But anyway I changed it till the other ioplacer is ready to use,
For the macros placement, I guess reason one might make more sense as the library merging doesn't really change anything in the library, just copy and paste. Is RePlAce placing macros successfully in the separate tests ?

tajayi, 2020-01-14:
Closing this as this implementation seems to have been abandoned

closed 2020-01-14 by tajayi