The-OpenROAD-Project/yosys #7: Structural verilog behavior

rajarshiroy, 2020-05-20:
I have a few questions about structural verilog:
1) If I want to specify standard cells in my design, should I use the (* blackbox *) syntax in my .v for the standard cells?
2) Would the physical synthesis process still take into account the blackboxed standard cell properties during timing analysis?
3) How would I allow yosys physical to resize the blackboxed standard cells as well?

Thanks.

ahmed-agiza, 2020-05-20:
Hi @rajarshiroy,
Physical synthesis in Yosys is now obsolete, the current version of Yosys is almost identical to YosysHQ/Yosys but with minor adjustments.
1. If you want to hardcode a standard cell library then yes you can provide the Verilog description for your standard cell (most standard cell libraries have Verilog description files) but in general, it is better to leave the standard cell mapping to Yosys/ABC unless there is a requirement to hardcode it (such as macros).
2. If you want to investigate buffering and resizing algorithms there is a bunch of them in [OpenPhySyn](https://github.com/scale-lab/OpenPhySyn/tree/develop/src/StandardTransforms) & [Resizer](https://github.com/The-OpenROAD-Project/OpenROAD/tree/openroad/src/resizer), they are mostly applied after placement.
3. Any tool will need the timing information (usually present in liberty files) of the standard the cell in order to do resizing. Resizer & some optimizations in OpenPhySyn can handle resizing if not done by the synthesis tool.

rajarshiroy, 2020-05-20:
Thank you @ahmed-agiza. That makes a lot of sense. Closing.

closed 2020-05-20 by rajarshiroy