The-OpenROAD-Project/yosys #8: ERROR: Can't open ABC output file `/tmp/yosys-abc-qguNOC/output.blif'

gkamendje, 2020-07-01:
Yosys 0.9+1706 (git sha1 4be891e8, gcc 7.5.0-3ubuntu1~18.04 -fPIC -Os)

I am getting the following error message while trying to synthesize my design.

ERROR: Can't open ABC output file `/tmp/yosys-abc-qguNOC/output.blif'
The file directory exists and write permissions are OK. There are no warnings that precede the error message. I get this error only when I run Yosys in the OpenROAD flow. I do not get the error while running Yosys in standalone mode with the same tech library and the synthesis script from the OpenRoad. So I guess that it is flow related but I don't know where to start digging. Here is an output of the ABC preceding the error message

```
38.1. Extracting gate netlist of module `\digital' to `/tmp/yosys-abc-qguNOC/input.blif'..
Extracted 8711 gates and 9201 wires to a netlist network with 490 inputs and 510 outputs.

38.1.1. Executing ABC.
Running ABC command: /home/myst/tmp_compile/OpenROAD-flow/tools/build/yosys/bin/yosys-abc -s -f /tmp/yosys-abc-qguNOC/abc.script 2>&1
ABC: ABC command line: "source /tmp/yosys-abc-qguNOC/abc.script".
ABC:
ABC: + read_blif /tmp/yosys-abc-qguNOC/input.blif
ABC: + read_lib -w /home/myst/tmp_compile/OpenROAD-flow/flow/./objects/my_tech/rad/merged.lib
ABC: Parsing finished successfully.  Parsing time =     2.86 sec
ABC: Scl_LibertyReadGenlib() skipped cell "BUSHOLDHD" without logic function.
ABC: Scl_LibertyReadGenlib() skipped cell "CLKLAHAQHDV0" without logic function.
ABC: Scl_LibertyReadGenlib() skipped cell "CLKLAHAQHDV1" without logic function.
```
--skipping many lines

`ABC: Scl_LibertyReadGenlib() skipped sequential ceERROR: Can't open ABC output file `/tmp/yosys-abc-qguNOC/output.blif'.`
Is there a switch I could enable to further debug this
Thanks

juspertor, 2020-12-21:
Same issue here: Executing ABC.ERROR: Can't open ABC output file `/tmp/yosys-abc-ihYbSB/output.blif'.  The issue only appears when running under Windows Subsystem for Linux (wsl, also with Ubuntu 180.04) same binaries on native Ubuntu 18.04 runs fine. Is there any workaround for this issue?

juspertor, 2020-12-23:
surprisingly this is an issue with the file name/file location. A call like 'abc -liberty "/mnt/c/Program Files (86)/mylib/mylib.lib" ' causes this error message while a call ''abc -liberty "/home/me/mylib.lib" ' runs fine.

maliberty, 2024-03-06:
Issues should be filed in the upstream rather than this fork

closed 2024-03-06 by maliberty